wokwi/wokwi-features #423: MAX7219 Custom Chip

LiamTHughes, 2022-09-25:
Hello,

I am trying to create an 8x8 LED Matrix display on Wokwi. As a challenge, I want to create the led matrix from scratch, with 64 individual LEDs. I have already created the LED matrix but I have realised that Wokwi does not have an individual MAX7219 chip - they only have the LED matrix and max2719 in one component. 

Is there a way to create a max2719 chip?

Thank you very much for your understanding and assistance.

M.

urish, 2022-09-25:
Hello! 

You can also create the MAX7219 using the custom chips API. 

However, note that there's a good reason Wokwi integrated them into a single part: performance. If these two parts are separate, the MAX7219 has to continuously scan the 8x8 LED matrix, and then then 8x8 LED matrix has to implement PWM (you can see how I create a [simple implementation](https://www.youtube.com/watch?v=7KYzPJ-tQno) here), which could be demanding on the CPU.

Most users just care about the end result (what they see on the matrix), so in order to give them the best user experience, the MAX7219 and the LED Matrix are integrated.

But please go ahead and create your own MAX7219 using the custom chip API - I'd love to see how it performs.

MarkWilliams12, 2022-09-25:
Hello Urish,

Thank you very much for the heads up. I like the challenge of creating my own MAX7219 chip but I think I may have bitten off more than I can chew with this one. Your video goes over my head! I tried to use two 74HC595 pieces to circumnavigate it, however, I seem to have reached a dead end with that one too. 

Thank you for your help nonetheless.

urish, 2022-09-25:
You're welcome!

I think - if you shared a link to your current work on the [discord chat server](https://wokwi.com/discord), someone might pick it up from there