wokwi/wokwi-features #426: Build failed for TinyTapeout Default Design: Missing Type Specifier

jdrosent, 2022-09-28:
**Describe the bug**
Wokwi TinyTapeout template fails to build when I press "Play" 

**To Reproduce**
Using the default TinyTapeout template (https://wokwi.com/projects/339800239192932947) and pressing play, the design fails. Error information is provided below.

Modifying and deleting components doesn't change the error.

**Expected behavior**
Pressing the "Play" button, I expected the TinyTapeout circuit to begin simulation.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Browser: Firefox

**Additional context**
Using other Wokwi simulations, like the Arduino Simon Game, no errors appear.

`/sketch/output-8-pins.chip.c:11:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(OUT0);
^
/sketch/output-8-pins.chip.c:11:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(OUT0);
           ^
/sketch/output-8-pins.chip.c:12:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(OUT1);
^
/sketch/output-8-pins.chip.c:12:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(OUT1);
           ^
/sketch/output-8-pins.chip.c:13:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(OUT2);
^
/sketch/output-8-pins.chip.c:13:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(OUT2);
           ^
/sketch/output-8-pins.chip.c:14:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(OUT3);
^
/sketch/output-8-pins.chip.c:14:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(OUT3);
           ^
/sketch/output-8-pins.chip.c:15:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(OUT4);
^
/sketch/output-8-pins.chip.c:15:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(OUT4);
           ^
/sketch/output-8-pins.chip.c:16:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(OUT5);
^
/sketch/output-8-pins.chip.c:16:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(OUT5);
           ^
/sketch/output-8-pins.chip.c:17:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(OUT6);
^
/sketch/output-8-pins.chip.c:17:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(OUT6);
           ^
/sketch/output-8-pins.chip.c:18:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(OUT7);
^
/sketch/output-8-pins.chip.c:18:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(OUT7);
           ^
/sketch/output-8-pins.chip.c:19:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(EXTOUT0);
^
/sketch/output-8-pins.chip.c:19:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(EXTOUT0);
           ^
/sketch/output-8-pins.chip.c:20:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(EXTOUT1);
^
/sketch/output-8-pins.chip.c:20:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(EXTOUT1);
           ^
/sketch/output-8-pins.chip.c:21:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(EXTOUT2);
^
/sketch/output-8-pins.chip.c:21:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(EXTOUT2);
           ^
/sketch/output-8-pins.chip.c:22:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(EXTOUT3);
^
/sketch/output-8-pins.chip.c:22:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(EXTOUT3);
           ^
/sketch/output-8-pins.chip.c:23:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(EXTOUT4);
^
/sketch/output-8-pins.chip.c:23:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(EXTOUT4);
           ^
/sketch/output-8-pins.chip.c:24:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(EXTOUT5);
^
/sketch/output-8-pins.chip.c:24:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(EXTOUT5);
           ^
/sketch/output-8-pins.chip.c:25:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(EXTOUT6);
^
/sketch/output-8-pins.chip.c:25:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(EXTOUT6);
           ^
/sketch/output-8-pins.chip.c:26:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
DEFINE_PIN(EXTOUT7);
^
/sketch/output-8-pins.chip.c:26:12: error: a parameter list without types is only allowed in a function definition
DEFINE_PIN(EXTOUT7);
           ^
/sketch/output-8-pins.chip.c:39:7: error: conflicting types for 'chip_init'
void* chip_init() {
      ^
/src/wokwi-api.h:45:54: note: previous declaration is here
extern __attribute__((export_name("chipInit"))) void chip_init(void);
                                                     ^
/sketch/output-8-pins.chip.c:61:29: warning: incompatible integer to pointer conversion passing 'int' to parameter of type 'pin_watch_config_t *' [-Wint-conversion]
    pin_watch(chip->ins[i], BOTH);
                            ^~~~
/src/wokwi-api.h:51:95: note: passing argument to parameter 'config' here
extern __attribute__((import_name("pinWatch"))) bool pin_watch(pin_t pin, pin_watch_config_t *config);
                                                                                              ^
/sketch/output-8-pins.chip.c:71:12: error: function cannot return function type 'void (void *, pin_t, uint32_t)' (aka 'void (void *, int, unsigned int)')
void EXPORT(chip_pin_change)(void *chip, pin_t pin, uint32_t value) {
           ^
/sketch/output-8-pins.chip.c:72:15: error: use of undeclared identifier 'chip'
  update_pins(chip);
              ^
17 warnings and 19 errors generated.`

urish, 2022-09-28:
Thanks for reporting! It will take a short while until we fix all the existing projects.

In the meanwhile, you can start from this link: https://wokwi.com/projects/344056279947280978

Or just copy the contents of `input-8-pins.chip.c` and `output-8-pins.chip.c` from the above project to your project.

urish, 2022-09-28:
All existing projects fixed, including the default TinyTapeout template.  Thanks again for reporting this!

closed 2022-09-28 by urish