wokwi/wokwi-features #427: Secrets Manager

chrisisbeef, 2022-09-28:
**Is your feature request related to a problem? Please describe.**
Storing secrets in your source code is a known major security issue. 

**Describe the solution you'd like**
Arduino has solved this problem in their web editor through the creation of a `Secrets` tab in the editor where those secrets are associated to the user and the project; and when a project is opened by a difference user, that association can not be made so the values in their secrets tab are empty and have to be filled in by them. In the code, a simple token replacement appears to be how it is implemented in the arduino cloud editor; anyplace a secret needs to be added in the code, it is referenced using the template `SECRET_YOUR_SECRET_NAME` and that value will be replaced by the secret string.

For example: 
If the `Secrets` tab contained the following secret
```
SECRET_SSID: Wokwi-Guest
```

Prior to compilation the template would go from this

```
const char SSID[] = SECRET_SSID;
```

to this
```
const char SSID[] = "Wokwi-Guest"
```

urish, 2022-09-28:
Thanks! Opened this request for [voting](https://wokwi.com/features?sort=newest#feature-1389957765)

Cloufish, 2023-03-10:
May I ask to clarify? o.O

I understand the idea - We want to not store secrets in plaintext inside the template
But @chrisisbeef described that:

> Prior to compilation the template would go from this
> 
> ```
> const char SSID[] = SECRET_SSID;
> ```
> 
> to this
> 
> ```
> const char SSID[] = "Wokwi-Guest"
> ```

How is it more secure If we have a plaintext secret? ;o
Did you type it by mistake the other way around @chrisisbeef ?

chrisisbeef, 2023-03-10:
> How is it more secure If we have a plaintext secret? ;o Did you type it by mistake the other way around @chrisisbeef ?

The secret exists only as Plaintext at at compile-time and outside of the IDE - the biggest benefit here is that it allows developers to quickly prototype an idea without having their secrets stored as part of the source code itself. The `SECRET_XXXXXX` token would be replaced during preprocessing by the actual value of the stored secret from a vault built in to the environment (there are a couple different ways to actually implement this) 

If you load up the Arduino IDE you can see an example of how they handle this problem

urish, 2024-06-30:
Closing this issue as it hasn't received any votes despite being open for almost two years.

closed 2024-06-30 by urish