wokwi/wokwi-features #428: Pan in and out compromised the text label size

matititam, 2022-10-06:
**Describe the bug**
The text (PIN label, IC name, and any other text present in the Wokwi parts in the simulation window) size was reduced. It is no more proportional to the Wokwi element
[image]

Link of the project: https://wokwi.com/projects/306031380875182657
[image]

**To Reproduce**
Not yet reproducible

**Expected behavior**
text size should always be proportional to the wokwi element size
**Environment (please complete the following information):**
 - OS: [Windows]
 - Browser [Chrome]
 - Version 105.0.5195.127

urish, 2022-10-06:
This seems to be a bug in Chrome - did you observe it in any other browsers?

matititam, 2022-10-13:
Just reporting again - Chrome browser, Windows 11
[image]

again during the zooming operation. Will try in other borders for a while

urish, 2022-11-27:
Found a way to reliably reproduce this in chrome:

1. Go to https://wokwi.com/projects/new/arduino-uno
2. Zoom in (or out) using the mouse wheel
3. Click on the Arduino Uno part to select it
4. Press Ctrl+A

The issue should reproduce.

urish, 2023-11-13:
This has been fixed

closed 2023-11-13 by urish