wokwi/wokwi-features #431: Support for upload Files for LittleFS or SPIFFS on esp

marelab, 2022-10-11:
It would be a great addition if it would possible to integrate the file upload for Spiffs  and LittleFS to the spiffs area of the esp. It is mostly overkill to do it over the external SD Card. But maybe a lot of the work for the SD card could be reused  just by adding a select flag to upload to SD or  SPIFF and integrate the upload tool for the esp in the buildchain when the flag is set. That would make a lot of fancy web apps possible that could host html img and css files.

urish, 2022-10-12:
Good idea! Opened for [voting](https://wokwi.com/features?sort=newest#feature-1405351032)

josephted, 2023-03-13:
Great idea. Essentials to manipulate SPIFF. I vote to do.

stefanbode, 2023-04-10:
I want to use the simulator to easily check different configurations of TASMOTA. With the ESP32 the file system on the ESP is an essential part for the configuration. Adding a SD card will not work in this case.

ShamWerks, 2023-10-24:
+1 for this feature. It would be awesome to have SPIFFS access.

dejavu1987, 2023-11-05:
+1 My very first project I wanted to try needs SPIFFS.

msn444, 2024-01-14:
I have no trouble using spiffsgen.py to create spiffs partitions that can be merged into the .bin file and mounted in Wokwi. The dealbreaker right now is that Wokwi doesn't seem to allow ESP32 code to write to the SPIFFS and have it persist to later sessions. Is this supported?

urish, 2024-01-14:
> The dealbreaker right now is that Wokwi doesn't seem to allow ESP32 code to write to the SPIFFS and have it persist to later sessions. Is this supported?

Not yet, but very likely to be supported in Wokwi for VS Code in the future. What's your use case for this?

msn444, 2024-01-15:
> Not yet, but very likely to be supported in Wokwi for VS Code in the future. What's your use case for this?
Storage of configuration and provisioning data, typically received via mqtt and then affecting device behaviour going forward. Do any practical IoT devices not do this?

More generally, I need *all* flash writes to persist between sessions, including to SPIFFS, NVS, OTA partitions, and the OTA data partition. These are all aspects of the system that would be really nice to be able to develop and test in a simulator. 

Ideally it would work like any other VM, where the mass storage device is emulated from an image file on the host and can be modified by the VM.

brentru, 2024-07-24:
+1 for this, maybe with LittleFS