wokwi/wokwi-features #434: Add breadboard element

urish, 2022-10-18:
a [Solderless breadboard](https://en.wikipedia.org/wiki/Breadboard) is a matrix of electrical connection points, in a plastic enclosure.

They come in various sizes. The common sizes are:

- Full: 60 rows, 800 points, including positive/negative bus strips
- Half: 30 rows, 400 points, including positive/negative bus strips
- Mini: 17 rows, 170 points, no bus strips

Ideally, our element will support at least these 3 standard configurations, selectable via the "size" attribute. e.g.:

```html
   <wokwi-breadboard size="full"></wokwi-breadboard>
```

## Full breadboard
[image]

## Half Breadboard
[image]

## Mini breadboard
[image]

urish, 2022-10-18:
Thanks to @AriellaE, the breadboard is now functional: https://wokwi.com/projects/328952970765926995

You still need to manually connect parts to the breadboard pins - placing an LED or a resistor on top of pins won't automatically connect them. But @AriellaE is on it!

demc, 2022-10-20:
> > Does this not work for Raspberry Pi Pico with MicroPython? I can't get it to work!
> 
> In general, the breadboard is still considered in beta. You can share a link to the project, and we'll try to help.

Wow thanks for the fast reply! 

https://wokwi.com/projects/346043917808960084

I can't figure out how to add the component to the palette. Do I need to enable beta mode?

urish, 2022-10-20:
> I can't figure out how to add the component to the palette. Do I need to enable beta mode?

Right now, the only way is to copy-paste it from a different project, e.g. this one https://wokwi.com/projects/328952970765926995

demc, 2022-10-20:
> > I can't figure out how to add the component to the palette. Do I need to enable beta mode?
> 
> Right now, the only way is to copy-paste it from a different project, e.g. this one https://wokwi.com/projects/328952970765926995

Hmm, I copy pasted into my project with no luck:

```
    { "type": "wokwi-breaboard", "id": "bb1", "top": 0, "left": 0, "attrs": {} },
```

tproffen, 2022-10-20:
When you paste in the line, make sure you have a valid JSON file - I had an
extra comma in the wrong place and nothing showed up. Once I fixed it it
showed up as expected.

-- 
*Dr. Thomas Proffen*
Facebook: tproffen <http://www.facebook.com/tproffen> | Twitter:
@ThomasProffen <https://twitter.com/ThomasProffen> | Github: tproffen
<https://github.com/tproffen> | LinkedIn: tproffen
<https://www.linkedin.com/in/tproffen/>

*Oak Ridge Computer Science Girls*
http://www.orcsgirls.org | Facebook: ORCSGirls
<https://www.facebook.com/orcsgirls> | Twitter: @ORCSGirls
<https://twitter.com/orcsgirls> | Donate  <http://orcsgirls.org/donate>

demc, 2022-10-20:
> When you paste in the line, make sure you have a valid JSON file - I had an extra comma in the wrong place and nothing showed up. Once I fixed it it showed up as expected.
> […](#)
> -- *Dr. Thomas Proffen* Facebook: tproffen <http://www.facebook.com/tproffen> | Twitter: @ThomasProffen <https://twitter.com/ThomasProffen> | Github: tproffen <https://github.com/tproffen> | LinkedIn: tproffen <https://www.linkedin.com/in/tproffen/> *Oak Ridge Computer Science Girls* http://www.orcsgirls.org | Facebook: ORCSGirls <https://www.facebook.com/orcsgirls> | Twitter: @orcsgirls <https://twitter.com/orcsgirls> | Donate <http://orcsgirls.org/donate>

Thanks, but I already checked that! The pane will not render if the JSON is invalid. The Pico and other components render but not the breadboard.

urish, 2022-10-20:
It should read `wokwi-breadboard` and not `wokwi-breaboard` (note the missing `d` in the middle).

But instead of copying the JSON itself, you can select the part in one project, Ctrl+C, go to the diagram of the other project, Ctrl+V and have the part copied. See [here](https://docs.wokwi.com/guides/diagram-editor#copying-and-pasting-parts) for more details

demc, 2022-10-20:
> It should read `wokwi-breadboard` and not `wokwi-breaboard` (note the missing `d` in the middle).
>
> But instead of copying the JSON itself, you can select the part in one project, Ctrl+C, go to the [...]

What an obvious mistake... thank you so much. I will make a donation now ;)

urish, 2022-11-15:
We have a beta of the breadboard. It works if you manually connect wires from the parts to the breadboard.

What doesn't work yet?
1. When placing an LED or pushbutton on top of the breadboard , it won't attach automatically to the breadboard. You need to manually connect the pins
2. Moving a breadboard does not move the parts placed on it (but you can shift-drag to select the breadboard and the parts on it to move them together)
3. If you add the breadboard after adding some components (e.g. LEDs), then the breadboard will go on top of the older components. One workaround is to manually edit diagram.json and move the wokwi-breadboard to the beginning of the part list.

Should we release what we have now and gradually add the missing features? Or are these show-stoppers and we should first implement them and then release the breadboard?

You can vote with reactions:  
🚀 to release what we have right now  
👀 to wait until we finish implementing the missing features

and feel free to elaborate in a comment!

tproffen, 2022-11-15:
Love the progress :) We are planning to use this for electronics classes. We will wait until the connection to the breadboard is automatic after placing a component. Adds too much confusion for the kids otherwise 😎

tyeth, 2022-11-17:
Agreed, could happily work around 2, but 1 and 3 are show-stoppers :)

urish, 2022-11-20:
Update: we added most of the missing breadboard functionality - parts now attach automatically to the breadboard, they move together with the breadboard, and when you add a breadboard, it goes behind the other parts.

Still missing: we need to fix an issue with the spacing between the top and bottom part of the breadboard. Also, rotating either the breadboard or parts interferes with the auto attach functionality.

urish, 2022-11-22:
The breadboard is now ready for mass consumption! 

https://wokwi.com/projects/328952970765926995

I'm keeping this issue open still as we are also working on half-breadboard and mini breadboard for you.

urish, 2022-11-23:
Half breadboard: https://wokwi.com/projects/349109488710582866

urish, 2022-11-23:
Mini breadboard: https://wokwi.com/projects/349124282187514451

and we're good to go!

closed 2022-11-23 by urish