wokwi/wokwi-features #435: ESP32-S2 PWM Frequency

Dlloydev, 2022-10-22:


urish, 2022-10-22:
Thanks for reporting! So the phases are fine, and the frequency is the only problem?

Dlloydev, 2022-10-22:
Yes, only the frequency.

I haven't done much testing with the phase shift as it uses a low level command and I don't expect it to be very accurate because the LEDC hardware timers don't synchronization.

urish, 2022-10-22:
Thanks! Have you had a chance to test it with the ESP32 too? Or only the S2?

Dlloydev, 2022-10-22:
Only the S2. My ESP32 is hiding somewhere. I guess lost hardware is a side effect of using simulation software!

urish, 2022-10-22:
> I guess lost hardware is a side effect of using simulation software!

Can I quote that? 😉

Dlloydev, 2022-10-22:
Sure!

urish, 2022-10-22:
Fix is up - can you please check now and report?

Dlloydev, 2022-10-22:
Wow, that was fast. Looks good!

[image]

closed 2022-10-22 by Dlloydev