wokwi/wokwi-features #438: Serial Monitor: Enable byte / hex read and write

samtal, 2022-10-24:
Currently, (afaik) the small serial emulation field enables to send and receive ascii codes only. 
Most serial monitoring programs (such as Hercules,  AccessPort, Eltima, etc.) enable to send and receive both ascii and hex. 
For Arduino app, it is mostly desired to use hex to enable transferring bytes or chars (That can easily be converted into ASCII, if required).
(I have tried to install the xterm and modify the serial to "terminal" but failed to get it to work).

urish, 2022-10-24:
Thanks for the feature request! Opened it for [voting](https://wokwi.com/features?sort=newest#feature-1420581404)

samtal, 2022-10-24:
To add to my ticket, on second thought, there is no need to develop more than **an interface to the standard RS232** that should enable to connect to **any commercially available serial port control**. This will open all possible features, eliminating the need for a  dedicated port control.

SashaSemenishchev, 2023-09-05:
This would be useful!

closed 2025-11-03 by urish