wokwi/wokwi-features #439: Custom Chip persistence

bmeisels, 2022-10-25:
Currently the Custom Chip feature doesn't support persistence (and although you can technicly load an initial value through the env variables this is kind of a hack).

I would like to see a persistence mechanism which would cover the following:
- support for persistence between simulation runs of the same project
-  Import and export of the data (when the simulation is not running) - this should allow for initializing the simulation with a data file
- A clear format for data files(JSON?) Which can be used by the simulator to load the data (possibly multiple options in the same file)
- Support for loading multiple initial data into the project and choosing one to load in to the simulation(when at rest)
- Persistence API in C for the custom Chip mirroring the above

Implementing the above would allow for persistent memory simulation (including custom Chip and https://github.com/wokwi/wokwi-features/issues/231)

Should we combine this request with https://github.com/wokwi/wokwi-features/issues/231 as they should probably be implemented using the same code?

urish, 2022-10-25:
Thanks! Any proposal for what the persistence API would look like?

We already have a basic buffer API (part of the [Framebuffer API](https://gist.github.com/urish/f9129d3d34ffd557eccf354907663051#framebuffer-api)), so if we only wanted to support binary data, this is one possible direction.

bmeisels, 2022-11-04:
So I would go for something similar but the persistent memory would be single dimensional unlike the 2-dimensional framebuffer.
Additionally the init function should get a named buffer and an argument if to always initialize the buffer or attempt to load an existing buffer. I can write a suggested interface if that would be helpful.

urish, 2022-11-04:
Yes please

Dlloydev, 2023-04-17:
Some thoughts:

Thinking of persistence for custom controls only, could this be done transparently?
If so, then no user intervention would be needed, other than perhaps turning a new "sticky controls" option on or off in diagram.json.

Right now its awesome to be able to duplicate a chip or copy a section of a diagram with a chip or chips, then paste in another project. The dependencies and part information in the project's diagram.json are automatically updated.

If sticky controls is enabled, then the part section in diagram.json would include the controls section from chip.json.
For example, if the scope chip has its controls pre-adjusted to suit the new project, then when copied/pasted to the new project, it would already have the appropriate settings for the new project. These are the new default settings unless changed by the user.

When just working on a single project and f sticky controls is enabled, then the part section of any chips used would include a controls section (from chip'json) and would store the latest settings, allowing any project to have unique settings.

If sticky controls is disabled, then the control defaults would be as initialized in the chip's code (current behavior).