wokwi/wokwi-features #440: [ESP32] taskYIELD()  does not work

dracunciliasis, 2022-10-25:
**Describe the bug**
When using taskYIELD() the processor hangs.

When using the code transcribed below it should be printing the following indefinitely:
Received = 100
Received = 200
Received = 100
Received = 200
Received = 100
Received = 200
...

instead it only prints once for each task and hangs:
Received = 100
Received = 200

**To Reproduce**
To reproduce the behavior, use the following code in ESP32 arduino project.

#include "Arduino.h"
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_task_wdt.h"

static void vSenderTask( void *pvParameters );
static void vReceiverTask( void *pvParameters );

QueueHandle_t xQueue;

/*-----------------------------------------------------------*/
void setup( void )
{
  vTaskPrioritySet(NULL, configMAX_PRIORITIES-1);
  Serial.begin(115200);
  xQueue = xQueueCreate( 5, sizeof( long ) );
  if( xQueue != NULL )
  {
	  xTaskCreatePinnedToCore( vSenderTask, "Sender1", 1024, ( void * ) 100, 1, NULL, 1);
	  xTaskCreatePinnedToCore( vSenderTask, "Sender2", 1024, ( void * ) 200, 1, NULL, 1);
	  xTaskCreatePinnedToCore( vReceiverTask, "Receiver", 1024, NULL, 2, NULL, 1);
  }
  else
  {
    /* The queue could not be created. */
  }
}
void loop()
{
	vTaskDelete( NULL );
}
/*-----------------------------------------------------------*/
static void vSenderTask( void *pvParameters )
{
long lValueToSend;
portBASE_TYPE xStatus;
  lValueToSend = ( long ) pvParameters;
  for( ;; )
  {
    xStatus = xQueueSendToBack( xQueue, &lValueToSend, 0 );
    if( xStatus != pdPASS )
      Serial.print( "Could not send to the queue.\r\n" );
    taskYIELD();
  }
}
/*-----------------------------------------------------------*/
static void vReceiverTask( void *pvParameters )
{
long lReceivedValue;
portBASE_TYPE xStatus;
const TickType_t xTicksToWait = 100 / portTICK_PERIOD_MS;
  for( ;; )
  {
    if( uxQueueMessagesWaiting( xQueue ) != 0 )
    {
    	Serial.print( "Queue should have been empty!\r\n" );
    }
    xStatus = xQueueReceive( xQueue, &lReceivedValue, xTicksToWait );
    if( xStatus == pdPASS )
    {
    	Serial.print( "Received = ");
    	Serial.println( lReceivedValue );
    }
    else
    {
    	Serial.print( "Could not receive from the queue.\r\n" );
    }
  }
}

**Expected behavior**
taskYIELD() is used to request a context switch to another task. However, if there are no other tasks at a higher or equal priority to the task that calls taskYIELD() then the RTOS scheduler will simply select the task that called taskYIELD() to run again.

**Environment (please complete the following information):**
 - ESP32
 - FreeRTOS

**Additional context**
Debugger is not available in ESP32, therefore cannot figure out where the problem is.
https://debugger.preview.wokwi.com/projects/346484762488078930

urish, 2022-10-25:
I can confirm this reproduces the behavior is different from an actual ESP32 chip.

output from running the same binary on the physical chip:

```
Received = 100
Received = 200
Received = 100
Received = 200
Received = 100
Received = 200
Received = 100
Received = 200
Received = 100
Queue should have been empty!
Received = 200
Queue should have been empty!
Received = 200
Queue should have been empty!
Received = 100
Queue should have been empty!
Received = 200
Queue should have been empty!
Received = 100
Could not send to the queue.
Could not send to the queue.
Received = 200
Received = 100
Received = 200
Received = 100
Queue should have been empty!
Received = 200
Queue should have been empty!
Received = 100
Queue should have been empty!
Received = 200
Queue should have been empty!
Received = 100
Queue should have been empty!
Received = 200
Could not send to the queue.
Could not send to the queue.
Received = 100
Received = 200
Received = 100
Received = 200
Received = 100
Queue should have been empty!
Received = 200
Queue should have been empty!
Received = 100
Queue should have been empty!
Received = 200
Queue should have been empty!
Received = 100
Queue should have been empty!
Received = 200
Could not send to the queue.
Could not send to the queue.
```

urish, 2022-11-02:
Initial investigation shows that the second core gets an exception in `vPortYield()`. I'll need further investigation to find the root cause.

urish, 2022-11-02:
Fix uploaded to wokwi.com!

Your can try it yourself at: https://wokwi.com/projects/346484762488078930

Note: please don't work with the debugger preview unless you need to actually test the new debugger, as we don't regularly merge fixes / new features into the debugger version.

closed 2022-11-02 by urish