wokwi/wokwi-features #441: Support deep sleep wakeup on ESP32

JeffinChang, 2022-10-31:
**Is your feature request related to a problem? Please describe.**
Attempt to run ESP32 example: TimerWakeUP.ino but it seems Wokwi, but there is wakeup when go to sleep. 

**Describe the solution you'd like**
Need support deep sleep wakeup on ESP32

urish, 2022-10-31:
Thanks for creating this feature request! I opened it for [voting](https://wokwi.com/features?sort=newest#feature-1429239393)

Phetdanai, 2023-12-22:
Good, @urish.

tyeth, 2024-02-01:
I'm interested in testing deep sleeping wifi based sensors, running arduino style sketches, something like the Adafruit WipperSnapper firmware. I'd be doing both browser and CI stuff, but eventually mostly just CI.
I'm also very keen to play with deep sleep on circuitpython which is more for rapid prototyping in the browser.

huseenhejazy7, 2024-03-17:
good