wokwi/wokwi-features #442: Able to change the pin names of the Logic Analizer

cheyao, 2022-11-03:
Being able to change the pin names of the Logic Analizer wold be a real nice feature so that you can know what the pins are for in logic analizer apps (like scansion)

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**

Being able to change the pin names

urish, 2022-11-04:
Added a "channelNames" attribute for this purpose: 

https://docs.wokwi.com/guides/logic-analyzer#customizing-the-channel-names

@Koepel mind updating your [Logic Analyzer example](https://wokwi.com/projects/325933824665977428) to use this new feature?

closed 2022-11-04 by urish

Koepel, 2022-11-04:
Added channel names and changed explanation for using the channel names.

urish, 2022-11-05:
Amazing, thanks!