wokwi/wokwi-features #444: Stepper motor limit switches

DarwinWasWrong, 2022-11-09:
when coding for stepper motors, it would be very handy for have some form of limit switches logic.

**Describe the solution you'd like**
either two switches on the motor rotation able to be positioned in the 360 degrees
or 
a attribute of switch 1 on/off at 3203 steps, switch 2 of/on at 10 steps

the you could code for like - home positioning or limits   interrupts

urish, 2022-11-15:
Thanks! opened this for [voting](https://wokwi.com/features?sort=newest#feature-1441305594)

drf5n, 2024-01-17:
You can simulate limit switches with a function in your code.  If, for example, you were using AccelStepper to drive the steppers, maybe something like this:

```
void simLimitSwitches() {
  const byte lowLimitPin = 13;
  const byte highLimitPin = 12;
  static long lastPos = 0;
  static bool initialized = false;
  if(initialized == false){
    pinMode(lowLimitPin,OUTPUT);
    pinMode(highLimitPin,OUTPUT);
    initialized = true;
  }
  long nowPos = stepper.currentPosition();
  if (nowPos != lastPos) {
    lastPos = nowPos;
    digitalWrite(lowLimitPin, nowPos < 30 ? HIGH : LOW);
    digitalWrite(highLimitPin, nowPos > 3000 ? HIGH : LOW);
  }
}
```

I dropped that function into https://wokwi.com/projects/327381547863769683 and called it inside loop(), and it toggles the LED_BUILTIN appropriately.