wokwi/wokwi-features #449: add south pole & north pole magnet & hall effect

rezkmattar, 2022-11-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

urish, 2022-11-15:
Thanks! Opened for [voting](https://wokwi.com/features?sort=newest#feature-1449278702)

urish, 2023-06-11:
Closing as this hasn't received any votes within the past 6 months

closed 2023-06-11 by urish

rezkmattar, 2023-06-22:
OK

-- 

Kind regards,

Rezk Mattar

Chairman Asst. For Technical

El-Massalla Co. (EMICA)

5 Waw, Khaled Taema st. Block 1224 Flat #6

Sheraton Heliopolis Zone, Cairo Egypt.

mob.: +20 (0) 12 22 11 47 59

tel.: +20 (0) 22 26 72 798, fax.: +20 (0) 222 26 72 799

email: ***@***.***