wokwi/wokwi-features #450: allow printing of code

rezkmattar, 2022-11-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

urish, 2022-11-15:
Do you mean printing to a printer?

urish, 2022-12-05:
Closing as there was no response. Feel free to add more information and reopen.

closed 2022-12-05 by urish

urish, 2022-12-07:
Thanks for explaining! Do you need to print code often?

One way to achieve it is to download the project as a .zip file:

[image]

And then open it locally and print the code

rezkmattar, 2022-12-07:
Thanks dear, yet it is quite lengthy.
Why can't I select all, then print?

-- 

Kind regards,

Rezk Mattar

Chairman Asst. For Technical

El-Massalla Co. (EMICA)

5 Waw, Khaled Taema st. Block 1224 Flat #6

Sheraton Heliopolis Zone, Cairo Egypt.

mob.: +20 (0) 12 22 11 47 59

tel.: +20 (0) 22 26 72 798, fax.: +20 (0) 222 26 72 799

email: ***@***.***

urish, 2022-12-08:
TBH we haven't anticipated anyone would want to print the code from the Wokwi interface.

Should I open this feature for voting?

rezkmattar, 2022-12-26:
I don't think so.
I just felt it is an easy improvement task & a good added value to the
great application, why not?
Thanks for your attention & good luck.

-- 

Kind regards,

Rezk Mattar

Chairman Asst. For Technical

El-Massalla Co. (EMICA)

5 Waw, Khaled Taema st. Block 1224 Flat #6

Sheraton Heliopolis Zone, Cairo Egypt.

mob.: +20 (0) 12 22 11 47 59

tel.: +20 (0) 22 26 72 798, fax.: +20 (0) 222 26 72 799

email: ***@***.***