wokwi/wokwi-features #451: MCP2515 CAN Bus

wastegate, 2022-11-17:
**Is your feature request related to a problem? Please describe.**
Not related to a problem.

**Describe the solution you'd like**
Add on board to emulate MCP2515 shield for CAN Bus communications.

urish, 2022-11-17:
Thanks, opened this for [voting](https://wokwi.com/features?sort=newest#feature-1452476943).

Can you please share more details about the use case? e.g. what kind of project you'd like to build with the MCP2515? Do you need other CAN bus parts?

urish, 2023-06-11:
Closing as this hasn't received any votes within the past 6 months

closed 2023-06-11 by urish

nshobe, 2024-09-24:
Well I'd vote for this. I want to develop multiple automotive related projects all of which require can bus to interact with existing systems.

Automotive Use Cases:
 - Data Logger
 - Smart system interaction ( if fuel pressure drop under power, trigger second fuel pump)
 - Static System interaction ( switch to secondary fuel system )
 - Extended Data Inputs ( brake pressure sensors, air temp before/after turbos, etc)

My point is that can bus is the standard communication for anything automotive and nearly every hobbyist is a tinkerer and once you've fully committed to go as far as removing the factory engine computer there's no "perfect" aftermarket solution and there's always some feature gap that something like an arduino could fit into.

There's even a github project for someone trying to make a full PDM (power distribution module) with current-sense amplifiers, and MOSFETs for a traditional fuse and relay replacement unit, ideally CAN bus integrated. Project uses a STM32 and MCP2515.