wokwi/wokwi-features #454: Possibility to connect real hardware

gorbounov, 2022-11-18:
It would be nice if it can enabled to connect a physical microcontroller (Arduino, and gradually others) to the WokWi. This would provide a great opportunity to:
- simulating the circuit in the stage of its design until the implementation of the main functionalities of the design entirely in the environment of the simulator
- **design verification by programming real hardware (calling avrdude could be used for programming a physical Arduino board)**
- building an environment for working in a remote laboratory

urish, 2022-11-18:
Thanks! I opened this for [voting](https://wokwi.com/features?sort=newest#feature-1454712413).

In the meanwhile, you can already download the compiled firmware .hex file from Wokwi, by pressing F1 in the code editor and selecting "Download Compiled HEX Firmware".

MarcelskyXD, 2022-12-27:
Possibly there could be a native program on the user's device which programs the code just like how they do it in the online arduino editor

aaronjamt, 2023-02-20:
I think using WebSerial and/or WebUSB could be a really cool option (at least for Chrome users since Mozilla has decided not to implement them in Firefox). ESPHome is one example of a program that does this: if you go to https://web.esphome.io/ on a Chrome browser, you can actually connect to an ESP32 from the browser without any software downloads. It works on any computer with Chrome installed, even Chromebooks. In fact, ESPHome can not only install firmware on the device, it can also show you the serial monitor right inside your browser.