wokwi/wokwi-features #457: Raspberry Pi Pico (SDK) missing hardware/pio.h (and others)

c1570, 2022-11-21:
**Describe the bug**
The Raspberry Pi Pico (SDK) project variant seems to be missing some Pico header files such as hardware/pio.h.
(note in the non-SDK project variant it works as expected)

**To Reproduce**
* Create a new Raspberry Pi Pico (SDK) project
* add `#include "hardware/pio.h"`
* compilation fails: `fatal error: hardware/pio.h: No such file or directory`

https://wokwi.com/projects/348954851689890388

**Expected behavior**
Compilation should not fail. :)

**Environment (please complete the following information):**
 - Browser Firefox latest

urish, 2022-11-21:
Thanks for reporting, we added the PIO library to the pico-sdk builder. Can you please test again now?

(If you immediately get an error about the missing header file, it might be the cache - try making some trivial change, like adding a comment and then recompile)

c1570, 2022-11-22:
Thanks a lot, it works now! Adding `hardware_clocks` (for clock_get_hz() etc.) would be great as well. Or is it possible to have some control over CMakeList.txt as there are a lot of possibly relevant core libraries?

urish, 2022-11-22:
Awesome! Added `hardware_clocks` as well.

Customizing CMakeLists.txt comes with a bunch of challenges (security & performance), so at the moment we'll just keep adding core libraries as users request them. 

There's also an upcoming Wokwi for VSCode Extension (#414), so that could be a better fit for users who require complete control over the build process.

p.s. what project are you working on?

closed 2022-11-22 by c1570