wokwi/wokwi-features #458: RP2040 PIO debugger: Breakpoints don't work occasionally

c1570, 2022-11-22:
**Describe the bug**
When running an RP2040 PIO program, some breakpoints (possibly on OUT only?) don't work.

**To Reproduce**
* https://wokwi.com/projects/348973734160237139
* Start simulation
* Open PIO debugger
* Set breakpoint on `out pins, 9 [10]`
* Observe breakpoint doesn't trigger (other lines trigger fine)

**Expected behavior**
* The breakpoint should trigger.

**Environment (please complete the following information):**
Firefox latest on Linux

urish, 2022-11-23:
Thanks for reporting. It seems like any breakpoint placed after a statement that waits for any external event (such as pulling from a queue) didn't work. Deployed a fix - can you please test again now?

c1570, 2022-11-23:
Awesome, this looks good now!

Possibly related - I noticed single stepping on the blocking PULL (instruction 27 in that project) doesn't do anything. Not sure if that's intentional - from a user perspective it's probably nicer to have the single step button wait until the PULL unblocks, then advance to the next instruction?

urish, 2022-11-29:
Thanks!

Well, right now the PIO debugger is not a priority for us. I agree that it'd be nicer to have the single step button wait until the PULL unblocks. We might revisit it after the debugger is in place (#205).

closed 2022-11-29 by urish