wokwi/wokwi-features #461: Reorganize "Add a new part" to have folders and subfolders (submenus)

shaos, 2022-11-24:
**Is your feature request related to a problem? Please describe.**
Right now drop-down menu of "Add a new part" became really long - I know search (filter) is helping, but it better be hierarchy like MCU boards in one submenu, descrete logic in other etc.

**Describe the solution you'd like**
Menu with submenues or dynamic menu that opens more options when user clicks to "parent" item in the top menu.

urish, 2022-11-24:
Thanks! Opened for [voting](https://wokwi.com/features?sort=newest#feature-1462837877)

urish, 2022-12-11:
@AriellaE is looking into this - she started a discussion on discord: https://discord.com/channels/787627282663211009/887330958875967520/1049711349330088037

urish, 2022-12-29:
And there we go - categories and also thumbnails:

[image]

Should we keep this issue open and wait for votes or is the current solution (dividing into categories) good enough?

shaos, 2022-12-30:
Looks better now, but still everything is open and listed as one huge list :)
At least categories must be hide-able (collapsable) and ideally all closed by default
and name of the category may have a number of components after it like "Basic (3)" (see attachement)
User may have setting somewhere what should be opened by default and it probably should be part of the project data like TT02+ template may have only Logic category fully opened (and may be Power) and when user copy this template project to their project this choice should be propagated as well (btw why buffer and flipflop with set/reset are not parts of the library?):
[image]

shaos, 2022-12-30:
So lets wait for votes :)

urish, 2022-12-30:
Thanks for the feedback! 

> why buffer and flipflop with set/reset are not parts of the library yet?

Buffer should be there. Flipflop with S/R  - basically waiting for @mattvenn to vet if it's ready for TT.