wokwi/wokwi-features #462: How to load a Spiffs or LittleFS partition in wokwi with esp32 for simulation.?

laldana73, 2022-11-24:
Hello.

there are some way to load a Spiffs or LittleFS partition in wokwi with esp32 for simulation.?

Thanks

urish, 2022-11-24:
Hello, if you build the binary on your computer, you can include an SPIFFS / LittleFS partition in the .bin file: https://docs.wokwi.com/guides/esp32#custom-application-bin-file

What project are you working on?

Also, in the future, [the discord community](https://wokwi.com/discord) is a better place for questions. There is a #questions channel there, where any questions about Wokwi are welcome.

closed 2022-11-24 by urish

laldana73, 2022-11-24:
Thanks, I am testing pikascript on esp32

urish, 2022-11-25:
You may need to add a "firmwareOffset" attribute to the "attrs" section of the esp32c3 chip in diagram.json, to tell it to load this .bin file at address 0:

```json
{
      "type": "board-esp32-c3-devkitm-1",
      "id": "esp",
      "top": -3.33,
      "left": -23.33,
      "attrs": { "firmwareOffset": "0" }
}
```

laldana73, 2022-11-25:
Perfect!! Is working Now
Thanks @urish

urish, 2022-11-25:
Awesome! I wasn't familiar with pikascript until now. Are you one of the developers?

laldana73, 2022-11-25:
No, i am developing a project based on pikascript.
I found it very interesting, and i hope be contributing soon