wokwi/wokwi-features #463: Compiler flags

ArduinoHocam, 2022-11-25:
Hi, Is there an option for the compiler flags that we can provide? For example, I'd like the change the optimization level for the avr-gcc compiler from -O0 to -O3 or vice versa. I know that I can do this on my local machine an upload the .hex but this is not a trivial solution where I am not on the machine I have access to Arduino related tools all the time etc. I believe this would be great option for all of us.

**Describe the solution you'd like**
Some of the usefull and simple compiler options (flags)  can be added to UI.

urish, 2022-11-25:
In general, the compiler flags go through the Arduino CLI. I think there's a way to change the optimization level for Arduino builds, but I'm not certain.

How would you determine which compiler options are valid? Do you have need for flags other than controlling the code optimization level?

ArduinoHocam, 2022-11-25:
> 
Hi, the compiler flags and corresponding options (normally) is located under the following folder for the Windows10, and IDE version greater than 1.6. `<exe>\Arduino\hardware\arduino\avr\platform.txt`

You are able to modify this file. For example the default template is as follows for me :
```

# Arduino AVR Core and platform.
# ------------------------------
#
# For more info:
# https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

name=Arduino AVR Boards
version=1.6.20

# AVR compile variables
# ---------------------

compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra

# Default "compiler.path" is correct, change only if you want to override the initial value
compiler.path={runtime.tools.avr-gcc.path}/bin/
compiler.c.cmd=avr-gcc
compiler.c.flags=-c -g -Os {compiler.warning_flags} -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects
compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections
compiler.c.elf.cmd=avr-gcc
compiler.S.flags=-c -g -x assembler-with-cpp -flto -MMD
compiler.cpp.cmd=avr-g++
compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto
compiler.ar.cmd=avr-gcc-ar
compiler.ar.flags=rcs
compiler.objcopy.cmd=avr-objcopy
compiler.objcopy.eep.flags=-O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0
compiler.elf2hex.flags=-O ihex -R .eeprom
compiler.elf2hex.cmd=avr-objcopy
compiler.ldflags=
compiler.size.cmd=avr-size

# This can be overridden in boards.txt
build.extra_flags=

# These can be overridden in platform.local.txt
compiler.c.extra_flags=
compiler.c.elf.extra_flags=
compiler.S.extra_flags=
compiler.cpp.extra_flags=
compiler.ar.extra_flags=
compiler.objcopy.eep.extra_flags=
compiler.elf2hex.extra_flags=

# AVR compile patterns
# --------------------

## Compile c files
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Compile S files
recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.S.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.S.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Create archives
# archive_file_path is needed for backwards compatibility with IDE 1.6.5 or older, IDE 1.6.6 or newer overrides this value
archive_file_path={build.path}/{archive_file}
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"
[... 186 lines cut ...]
tools.avrdude.erase.params.verbose=-v
tools.avrdude.erase.params.quiet=-q -q
tools.avrdude.erase.pattern="{cmd.path}" "-C{config.path}" {erase.verbose} -p{build.mcu} -c{protocol} {program.extra_params} -e -Ulock:w:{bootloader.unlock_bits}:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:{bootloader.high_fuses}:m -Ulfuse:w:{bootloader.low_fuses}:m

tools.avrdude.bootloader.params.verbose=-v
tools.avrdude.bootloader.params.quiet=-q -q
tools.avrdude.bootloader.pattern="{cmd.path}" "-C{config.path}" {bootloader.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{runtime.platform.path}/bootloaders/{bootloader.file}:i" -Ulock:w:{bootloader.lock_bits}:m

tools.avrdude_remote.upload.pattern=/usr/bin/run-avrdude /tmp/sketch.hex {upload.verbose} -p{build.mcu}

tools.avrdude.upload.network_pattern="{network_cmd}" -address {serial.port} -port {upload.network.port} -sketch "{build.path}/{build.project_name}.hex" -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return}

# USB Default Flags
# Default blank usb manufacturer will be filled in at compile time
# - from numeric vendor ID, set to Unknown otherwise
build.usb_manufacturer="Unknown"
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'

```

I'm sorry that I don't know what wokwi doing behind the scenes but I thought that, we can simply create a list of valid options, parse from the UI, if they are valid (ex: no typo, or valid for avr gcc compiler) pass it to the backend. 

In the IDE, there is an option for the verbose level. Also -W is also helpful.

If this idea suitable for the wokwi backend (I didn't check the implementation to be honest) I think we can add this as a cool feature ?

urish, 2022-12-05:
Thanks for the detailed comment!

I've looked into this, and this would be to complex to support for the Wokwi backend. However, users who need to customize their build environment still have options:

1. Build a .hex file locally and load it into Wokwi
2. Use a cloud development environment such as Gitpod or Github Codespaces
3. Soon, there'll be a Visual Studio Code plugin Wokwi (#414), so users will be able to use any local tool for their builds (e.g. Arduino CLI, Platform IO, etc.), and customize the builds as they like.

closed 2022-12-05 by urish

sutaburosu, 2023-01-31:
> Is there an option for the compiler flags that we can provide?

You can change the optimization level from within the source code with GCC `#pragma`s:

```cpp
#pragma GCC optimize "-O2"
```

You can even set different optimization levels for each function if you want. [This sketch shows that technique](https://wokwi.com/projects/318934180914266706).

urish, 2023-02-05:
> You can even set different optimization levels for each function if you want. [This sketch shows that technique](https://wokwi.com/projects/318934180914266706).

TIL