wokwi/wokwi-features #464: Breadboard alignment

TimMathias, 2022-11-26:
The breadboard has some translation and scaling issues where the text becomes mis-aligned.

[image]

This manifests after zooming in and out.

 - Windows 10 22H2 (64-bit)
 - Chrome
 - Version 107.0.5304.107 (Official Build) (64-bit)

The original SVG contains translation and scaling commands which the browser seems to be misinterpreting.

The following SVG is for testing and has been tweaked to replace the relative coordinates with absolute coordinates:

```svg
<svg x="0in" y="0in" width="6.8in" height="2.3in" viewBox="0 0 6800 2300" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink">
    <defs>
        <pattern id="Sockets" patternUnits="userSpaceOnUse" x="-20" y="-20" width="100" height="100">
            <rect x="0" y="0" width="40" height="40" fill="#777" />
        </pattern>
        <pattern id="SocketsSpaced" patternUnits="userSpaceOnUse" x="480" y="-20" width="600" height="100">
            <rect x="0" y="0" width="40" height="40" fill="#777" />
            <rect x="100" y="0" width="40" height="40" fill="#777" />
            <rect x="200" y="0" width="40" height="40" fill="#777" />
            <rect x="300" y="0" width="40" height="40" fill="#777" />
            <rect x="400" y="0" width="40" height="40" fill="#777" />
        </pattern>
    </defs>
    <rect x="0" y="0" width="6800" height="2300" rx="30" stroke="#000000" stroke-width="1" fill="#EEEFED" />
    <rect x="480" y="180" width="5890" height="190" stroke="#000000" stroke-width="0" fill="url(#SocketsSpaced)" />
    <rect x="280" y="580" width="6290" height="490" stroke="#000000" stroke-width="0" fill="url(#Sockets)" />
    <rect x="100" y="1110" width="6600" height="80" stroke="#000000" stroke-width="0" fill="#E3E3E3"></rect>
    <rect x="280" y="1280" width="6290" height="490" stroke="#000000" stroke-width="0" fill="url(#Sockets)" />
    <rect x="480" y="1980" width="5890" height="190" stroke="#000000" stroke-width="0" fill="url(#SocketsSpaced)" />
    <line x1="480" y1="100" x2="6320" y2="100" stroke="#F3B1A9" stroke-width="10" />
    <line x1="480" y1="400" x2="6320" y2="400" stroke="#A0E0F5" stroke-width="10" />
    <line x1="480" y1="1900" x2="6320" y2="1900" stroke="#F3B1A9" stroke-width="10" />
    <line x1="480" y1="2200" x2="6320" y2="2200" stroke="#A0E0F5" stroke-width="10" />
    <g font-family="Tahoma, sans-serif" font-size="100" font-weight="400">
        <text x="400" y="131" text-anchor="middle" style="fill: #F3B1A9">+</text>
        <text x="400" y="431" text-anchor="middle" style="fill: #A0E0F5">–</text>
        <text x="6400" y="131" text-anchor="middle" style="fill: #F3B1A9">+</text>
        <text x="6400" y="431" text-anchor="middle" style="fill: #A0E0F5">–</text>
        <text x="400" y="1931" text-anchor="middle" style="fill: #F3B1A9">+</text>
        <text x="400" y="2231" text-anchor="middle" style="fill: #A0E0F5">–</text>
        <text x="6400" y="1931" text-anchor="middle" style="fill: #F3B1A9">+</text>
        <text x="6400" y="2231" text-anchor="middle" style="fill: #A0E0F5">–</text>
    </g>
    <g font-family="Tahoma, sans-serif" font-size="80" font-weight="400" style="fill: #777">
        <text x="200" y="618" text-anchor="middle">a</text>
        <text x="200" y="718" text-anchor="middle">b</text>
        <text x="200" y="818" text-anchor="middle">c</text>
        <text x="200" y="918" text-anchor="middle">d</text>
        <text x="200" y="1018" text-anchor="middle">e</text>
        <text x="200" y="1318" text-anchor="middle">f</text>
        <text x="200" y="1418" text-anchor="middle">g</text>
        <text x="200" y="1518" text-anchor="middle">h</text>
        <text x="200" y="1618" text-anchor="middle">i</text>
        <text x="200" y="1718" text-anchor="middle">j</text>
        <text x="6600" y="618" text-anchor="middle">a</text>
        <text x="6600" y="718" text-anchor="middle">b</text>
        <text x="6600" y="818" text-anchor="middle">c</text>
        <text x="6600" y="918" text-anchor="middle">d</text>
        <text x="6600" y="1018" text-anchor="middle">e</text>
        <text x="6600" y="1318" text-anchor="middle">f</text>
        <text x="6600" y="1418" text-anchor="middle">g</text>
        <text x="6600" y="1518" text-anchor="middle">h</text>
        <text x="6600" y="1618" text-anchor="middle">i</text>
        <text x="6600" y="1718" text-anchor="middle">j</text>
        <text x="300" y="530" text-anchor="middle">1</text>
        <text x="700" y="530" text-anchor="middle">5</text>
        <text x="1200" y="530" text-anchor="middle">10</text>
        <text x="1700" y="530" text-anchor="middle">15</text>
        <text x="2200" y="530" text-anchor="middle">20</text>
        <text x="2700" y="530" text-anchor="middle">25</text>
        <text x="3200" y="530" text-anchor="middle">30</text>
        <text x="3700" y="530" text-anchor="middle">35</text>
        <text x="4200" y="530" text-anchor="middle">40</text>
        <text x="4700" y="530" text-anchor="middle">45</text>
        <text x="5200" y="530" text-anchor="middle">50</text>
        <text x="5700" y="530" text-anchor="middle">55</text>
        <text x="6200" y="530" text-anchor="middle">60</text>
        <text x="300" y="1818" text-anchor="middle">1</text>
        <text x="700" y="1818" text-anchor="middle">5</text>
        <text x="1200" y="1818" text-anchor="middle">10</text>
        <text x="1700" y="1818" text-anchor="middle">15</text>
        <text x="2200" y="1818" text-anchor="middle">20</text>
        <text x="2700" y="1818" text-anchor="middle">25</text>
        <text x="3200" y="1818" text-anchor="middle">30</text>
        <text x="3700" y="1818" text-anchor="middle">35</text>
        <text x="4200" y="1818" text-anchor="middle">40</text>
        <text x="4700" y="1818" text-anchor="middle">45</text>
        <text x="5200" y="1818" text-anchor="middle">50</text>
        <text x="5700" y="1818" text-anchor="middle">55</text>
        <text x="6200" y="1818" text-anchor="middle">60</text>
    </g>
</svg>
```

It produces the following layout which is much the same as the original:

[image]

urish, 2022-11-26:
Thanks Tim! 

Misaligned texts only happens on Chrome? or other browsers as well?

Did you find a consistent way to reproduce it? #428 may also be related

TimMathias, 2022-11-27:
It happens on Chrome and Edge. That's all I've tested. It is often (but not always) repeatable when zooming in then back out again. It seems that the browser loses track of the transforms.

urish, 2022-11-28:
Thanks! Deployed a workaround (setting the `text-rendering` CSS property to "geometricPrecision").

Can you please check if this still happens?

TimMathias, 2022-11-28:
That seems to have fixed it in Chrome and Edge.

guillaume-dorczynski, 2022-11-29:
Hello, sorry I'm late to the breadboard party, not much free time... This is what I had in mind
```
<svg width="6.8in" height="2.3in" viewBox="0 0 6800 2300" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink">
	<defs>
		<style>.text { font-family: Tahoma, sans-serif; font-size: 80px; font-weight: 600; text-anchor: middle; dominant-baseline: middle }</style>
		<pattern id="hole" patternUnits="userSpaceOnUse" x="-20" y="-20" width="100" height="100">
			<rect width="40" height="40" fill="#777"/>
		</pattern>
		<rect id="rows" x="280" y="580" width="6240" height="440" fill="url(#hole)"/>
		<g id="letters">
			<text y="600">A</text>
			<text y="700">B</text>
			<text y="800">C</text>
			<text y="900">D</text>
			<text y="1000">E</text>
			<text y="1300">F</text>
			<text y="1400">G</text>
			<text y="1500">H</text>
			<text y="1600">I</text>
			<text y="1700">J</text>
		</g>
		<g id="numbers">
			<text x="600">60</text>
			<text x="1100">55</text>
			<text x="1600">50</text>
			<text x="2100">45</text>
			<text x="2600">40</text>
			<text x="3100">35</text>
			<text x="3600">30</text>
			<text x="4100">25</text>
			<text x="4600">20</text>
			<text x="5100">15</text>
			<text x="5600">10</text>
			<text x="6100">5</text>
			<text x="6500">1</text>
		</g>
		<g id="bus_half" stroke-width="10">
			<rect x="280" y="180" width="3040" height="140" fill="url(#hole)"/>
			<line x1="280" x2="3320" y1="100" y2="100" stroke="#F3B1A9"/>
			<line x1="280" x2="3320" y1="400" y2="400" stroke="#A0E0F5"/>
		</g>
		<g id="signs">
			<text x="200" y="105" fill="#F3B1A9">+</text>
			<text x="200" y="405" fill="#A0E0F5">–</text>
		</g>
		<g id="bus" stroke-width="10">
			<use href="#bus_half"/>
			<use href="#bus_half" transform="translate(3200,0)"/>
			<g class="text" style="font-size:100px">
				<use href="#signs"/>
				<use href="#signs" transform="translate(6400,0)"/>
			</g>
		</g>
	</defs>
	<rect width="6800" height="2300" rx="30" fill="#EEEFED"/>
	<use href="#bus"/>
	<use href="#rows"/>
	<rect x="100" y="1110" width="6600" height="80" fill="#E3E3E3"/>
	<use href="#rows" transform="translate(0,700)"/>
	<use href="#bus" transform="translate(0,1800)"/>
	<g class="text" fill="#777">
		<use href="#letters" x="200"/>
		<use href="#letters" x="6600"/>
		<use href="#numbers" y="520"/>
		<use href="#numbers" y="1810"/>
	</g>
</svg>
```
[image]

urish, 2022-11-29:
@guillaume-dorczynski thanks! Seems like your breadboard variation has split power rails. Or did I miss something?

guillaume-dorczynski, 2022-11-29:
> @guillaume-dorczynski thanks! Seems like your breadboard variation has split power rails. Or did I miss something?

Yes, I think it's better, in case we could have different voltages in Wokwi.. :)

AriellaE, 2022-11-29:
Hello @guillaume-dorczynski , Thank's for your feedback. Can you please open a [new issue](https://wokwi.com/features) for your request?

closed 2022-11-29 by AriellaE