wokwi/wokwi-features #466: add pi pico w

RakibRyan, 2022-11-30:
add support for a wifi based board

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

urish, 2022-11-30:
Thanks! Opened this request for [voting](https://wokwi.com/features?sort=newest#feature-1469127224)

urish, 2023-01-21:
Some initial findings:

The WiFi (+Bluetooth 5.2) chip is Infineon CYW43439, and it is connected to the RP2040 over SPI. The pin assignment is as follows:

| RP2040 Pin | CYW43439 Pin | Description              |
| ---------- | ------------ | ------------------------ |
| `GPIO23`   | `WL_ON`      | Wireless power on signal |
| `GPIO24`   | `WL_D`       | wireless SPI data/IRQ    |
| `GPIO25`   | `WL_CS`      | wireless SPI CS \*       |
| `GPIO29`   | `WL_CLK`     | Wireless SPI CLK \*      |

\* When `WL_CS` is high, GPIO29 is connected to VSYS (so you can read VSYS via ADC3 input)

Relevant schematic section:

[image]

urish, 2023-03-27:
This feature has been getting a large number of votes recently, so I ordered a Pi Pico W to prepare for implementation.

urish, 2023-03-29:
Found a blog post with tons of useful information: https://iosoft.blog/2022/12/06/picowi/

urish, 2023-04-06:
Initial implementation of the Pi Pico W is live on Wokwi!

Here are some examples:

- [Arduino](https://wokwi.com/projects/360610135224442881)
- [MicroPython](https://wokwi.com/projects/360519097147837441)
- [Pi Pico SDK](https://wokwi.com/projects/360480722185134081)

documentation and more information coming over the next few weeks.

d3lta-v, 2023-05-08:
Just curious, would the Wokwi team consider implementing a CircuitPython version for the RPi Pico W? Since there is already a CircuitPython version for the Pico.

urish, 2023-05-14:
@d3lta-v not a priority for the first release. We aim for Pico SDK and micropython compatibility, since they are both officially supported by the Pi foundation, and Arduino compatibility since Arduino is the most popular platform on Wokwi.

PPATSSAR, 2023-07-24:
Could you please have a look into connecting a Pico W to a NTP time server using MicroPython:

import ntptime
ntptime.host = "pool.ntp.org"
ntptime.settime()

Frowsty, 2023-09-19:
After some testing with the Pico-W beta board I and Koepel (discord user) has found out that it's currently unable to connect to WiFi when test example is re-compiled meaning only the pre-compiled test example works as of now.

The current testing has included the following steps.
1. Make a project from scratch with the pico-w beta template, including as well as excluding the cw43 attribute for the board
2. Modifying the existing WiFi example

urish, 2025-03-29:
Closing as we already have the Pi Pico W in Wokwi

closed 2025-03-29 by urish