wokwi/wokwi-features #467: Request for 3-swith DIP module (from FB group for HERO/Arduino project)

Chappie78, 2022-11-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I spoke to someone from Wokwi in the Hero Project/Arduino group after finding your simulator (AWESOME, btw) and asked for a 3-switch DIP - you only have an 8-switch DIP in the item list. They suggested I post it here.

Also, while placing the 8-switch DIP on the simulator, both rows don't connect, I had to connect the top row by drawing a 'wire' from the top connections to a space on the breadboard. Am I doing something wrong or is this a bug?

urish, 2022-11-30:
Thanks! I opened this request for [voting](https://wokwi.com/features?sort=newest#feature-1469319677)

> Also, while placing the 8-switch DIP on the simulator, both rows don't connect, I had to connect the top row by drawing a 'wire' from the top connections to a space on the breadboard. Am I doing something wrong or is this a bug?

It sounds like a bug indeed. Can you please [open a separate bug report](https://github.com/wokwi/wokwi-features/issues/new/choose) for this?

urish, 2022-11-30:
Update: we already fixed the issue with the 8-DIP switch + breadboard, no need to open a separate report.

urish, 2023-06-11:
Closing as this hasn't received any votes within the past 6 months

closed 2023-06-11 by urish