wokwi/wokwi-features #468: Bug report - 8-switch DIP module does not connect on top row when placed on Breadboard

Chappie78, 2022-11-30:
**Describe the bug**
While placing the 8-switch DIP on the simulator, both rows don't connect, I had to connect the top row by drawing a 'wire' from the top connections to a space on the breadboard

**To Reproduce**
Just place the DIP switch on the breadboard

**Expected behavior**
When placing the switch, I expect all the connections to go green to show they are 'in', as per the bottom row.

**Environment (please complete the following information):**
 - Mac OS (latest)
 - Browser = Chrome
 - Version 107.0.5304.110

**Additional context**
See attached image.

[image]

urish, 2022-11-30:
Thanks for the detailed report.

Fixed! Can you please test now

Chappie78, 2022-11-30:
Magic! Looks like it's working now thanks :)

[image]

closed 2022-11-30 by urish