wokwi/wokwi-features #469: Raspberry Pico Rust support

passcod, 2022-12-01:
Like #334, but for the Pico!

urish, 2022-12-01:
Thanks for the suggestion!

Is this something you'd like to work on? Is there a big community around Rust + Pi Pico?

passcod, 2022-12-01:
Not nearly as much as ESP, but there's a decent community around Rust and RP2040 (and Pico by extension). The original post in #334 mentioned Rust/Pico too! I was just passing by and thought that was a curious omission given [the rust page](https://wokwi.com/rust) and [the pico sketch](https://wokwi.com/projects/new/pi-pico?template=pi-pico); I'm not (yet?) a wokwi user.

urish, 2022-12-01:
Thanks! 

With ESP32, the Espressif Rust Team has been really helpful putting together all the example projects, etc.

I've opened this for [voting](https://wokwi.com/features?sort=newest#feature-1470657750), even though - this would probably move forward much faster if there was someone who wanted to work on it and push it forward.

urish, 2023-06-11:
Closing, as this hasn't received any votes in the past 6 months

closed 2023-06-11 by urish

Koepel, 2024-01-04:
There are regularly requests for Rust on the Raspberry Pi Pico in the Discord channel.

sturmenta, 2024-04-07:
I would really love to try Rust on a Raspberry Pi Pico using your app ❤️