wokwi/wokwi-features #472: Add an ifrared LED

usernameeReal, 2022-12-03:
Pretty self-explanatory.
It should be able to transmit signals receivable with the IR receiver component.

urish, 2022-12-03:
Thanks! Opened this request for [voting](https://wokwi.com/features?sort=newest#feature-1474080854)

freemansoft, 2023-02-06:
How is this different than any other LED?   Do you mean you want an IR transmitter/receiver pair?  That is the component that would let you communicate over IR.  The IR LED isn't going to do anything by itself in a simulation.

aaronjamt, 2023-02-20:
> Pretty self-explanatory. It should be able to transmit signals receivable with the IR receiver component.

Couldn't you just connect a wire from the pin you *would* connect to an IR LED to the pin you have connected to the IR Receiver for the same effect?

> How is this different than any other LED? Do you mean you want an IR transmitter/receiver pair? That is the component that would let you communicate over IR. The IR LED isn't going to do anything by itself in a simulation.

There's already an IR Receiver component that can be used with the IR Remote component. OP is just asking to add an IR LED to transmit signals which can be picked up by the IR Receiver component.

freemansoft, 2023-02-20:
So the request is to have 

- a functioning IR transmitter/receiver pair.