wokwi/wokwi-features #481: Pico on breadboard width mismatch and not connecting to both sides

tproffen, 2022-12-08:
Love the new breadboard feature. It works great, except when I added the Raspberry Pico (CircuitPython), the distance of the pins does not match the breadboard and only one side auto connects to the breadboard pins and wires need to be run to the Pico pins on the not connected side (see https://wokwi.com/projects/350444482715451988).

AriellaE, 2022-12-08:
Hello @tproffen, 

I'm very happy to hear that you love the breadboard!
I fixed the Raspberry Pico and it should be fine now. 

Thank you for helping us make Wokwi better.

closed 2022-12-08 by AriellaE

tproffen, 2022-12-08:
@AriellaE Thank you so much - this must have been the quickest bug fix I have ever seen 🥰 There is one nore thing 😨 It looks like the Pico being one row to wide as you can see on the screenshot vs. picture from the actual device. So you loose one row on one side currently. 

[image]

[image]