wokwi/wokwi-features #482: Analog API pin_dac_write output cannot drive RGB led

ancebfer, 2022-12-11:
**Describe the bug**
After adding a RGB led (common cathode) to custom-chip-analog example, the RGB led never lights for any voltage output.

**To Reproduce**
Modified custom-chip-analog example:
https://wokwi.com/projects/330112801381024338

**Expected behavior**
RGB led light intensity should change proportional to voltage as in rgb-slides example:
https://wokwi.com/projects/306455554559050306

**Environment (please complete the following information):**
 - Debian stable
 - Firefox ESR

ancebfer, 2022-12-11:
Right modified custom-chip-analog-rgb example link:
https://wokwi.com/projects/350784647934771795

urish, 2022-12-12:
Yeah, unfortunately, LEDs (all of them, RGB included) do not really support analog inputs at the moment. A workaround is to use another custom chip which would do analog to PWM conversion.

ikostan, 2024-06-19:
Hi @urish,
Is this why RGB led doesn't work in my [WOKWI project](https://wokwi.com/projects/401004574253872129)?