wokwi/wokwi-features #487: Pagination fails for some projects and makers' pages

sutaburosu, 2022-12-19:
When viewing [my projects](https://wokwi.com/dashboard/projects) I can see the first page fine, but scrolling down never loads subsequent pages.

The responses take several seconds and the result is "500: Internal server error".
[image]

This seems to affect some makers' pages too. [Your own maker page](https://wokwi.com/makers/urish) loads all pages fine. [My maker page](https://wokwi.com/makers/sutaburosu) fails to load past the first page.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Browser:  Chrome v108.0.5359.125, Firefox v108.0.1

PS Apologies for my extended absence from the Discord. I hope to have more free time next year.

urish, 2022-12-19:
Steve! 

Missed you so much, great to hear again from you :)

Looking into this right way

urish, 2022-12-19:
Fix is up! Mind trying now and reporting?

sutaburosu, 2022-12-19:
That works perfectly now. Thanks, Uri.

closed 2022-12-19 by sutaburosu