wokwi/wokwi-features #488: How do I open a bi-directional terminal for IoT Basic?

twoporylyj, 2022-12-20:
**Describe the bug**
When I compile and run Stefan's IoT Basic, i get only an unidirectional terminal, which is not interactive, and I cannot tell the Basic interpreter to do something, all I get is a startup prompt...

**To Reproduce**
Open a blank page with e.g. Nano in the emulator.
Remove file with default sketch.
Upload all these four files into the emulator from here:
https://github.com/slviajero/tinybasic/tree/main/TinybasicArduino
Start emulation and wait for terminal to appear.

**Expected behavior**
Bi-directional terminal instead of unidirectional one. Ability to communicate with Basic interpreter beyond the startup prompt.

**Environment (please complete the following information):**
 Chromium based browser on Android 9

**Additional context**
None

urish, 2022-12-20:
You can change the Serial Monitor into a Serial Terminal by setting the [display mode](https://docs.wokwi.com/guides/serial-monitor#display) to "terminal" in diagram.json:

```json
  "serialMonitor": { "display": "terminal" }
```

Here is a complete example (which you can also use as a starting point):

https://wokwi.com/projects/308893120796295745

Good luck running IoT Basic! Please do share the project on our [discord community](https://wokwi.com/discord). There's a #projects channel for sharing.

closed 2022-12-20 by urish

twoporylyj, 2022-12-21:
Thanks, it was so simple!

urish, 2022-12-21:
You're welcome :-)

twoporylyj, 2022-12-22:
To all experimenters, if you still experience problems after this change (interpreter does not respond to what user types), take hardware-arduino.h file, find there the following line:
#undef USESPICOSERIAL
And change it as shown below:
#define USESPICOSERIAL 
And everything becomes excellent.

twoporylyj, 2022-12-23:
Just programmed an example. Four spaces before each line are required to correctly automatically paste text with line numbers from Android clipboard to Javascript terminal.
    10 pinm 2,2
    20 pinm 3,2
    30 pinm 11,2
    35 pinm 12,2
    40 print &2,dread(2);
    50 print &2,dread(3);
    60 print &2,dread(11);
    70 print &2
    80 delay 100
    75 if dread(12)=0 goto 100
    90 goto 40
    100 print &2, "Stopped"
    110 stop
This program simulates behaviour of some PLCs which display state of all inputs on LCD. In this case, there are four pulled-up inputs connected to buttons. State of three of them is displayed on the HD44780 compatible display. When none is pressed, there is 111. If you push one of the buttons, its state changes to 0. Like this: 011. Multitouch is supported by the emulator, i tried to push all three buttons simultaneously, and it works. The fourth button stops the program with displaying "Stopped".