wokwi/wokwi-features #489: the text selection

disappear123df, 2022-12-22:
****im using l tablet to code but every time I try to select a word I cant and also texting is a little bit hard it I need to Tap and hold For keyboard to Show

**Describe the solution you'd like**
----> if it's possible you can 1 tap show the keyboard 2 tap select the text I like the tap and hold feature I don't want to sacrifice that

disappear123df, 2023-01-13:
also you can add a select to the pop-up menu when you left click

urish, 2023-01-15:
Thanks for reporting the issue! What tablet is it?

Unfortunately, the code editor we're using, Monaco Editor, does not provide a good editing experience on tablets and mobile devices. There's an open issue (#208) about using a different code editor with better tablet support, but unfortunately, this issue get only [a few votes](https://wokwi.com/features#feature-1028025160) so far.

disappear123df, 2023-01-15:
I'll start voting next month