wokwi/wokwi-features #492: Backspace does not work as expected in the editor

JM-FRANCE, 2022-12-28:
When using wokwi on an
iPad  within Safari you can’t easily select all text to remove the template code that is there in every project.  So you have to position the cursor at the end of the code and keep the virtual backspace key pressed down until everything is removed.  

the typical behavior of backspace is to remove one char at a time but if you keep it pressed down the repetition pace accelerates which is good and after a while instead of events to delete one char you get events to delete a full word to go even faster. That’s where wokwi’s text editor fails, instead of deleting a full word and keeping the acceleration it goes back to the snail pace of one char at a time. 

**Expected behavior**
- Implement backspace delete world event 
- Implement better text selection on tablet (it’s impossible to select anything) and standard tablet gestures to move the cursor around

urish, 2022-12-28:
Don't have an iPad at end, but can you please test if the same happens with monaco-editor in this page? https://microsoft.github.io/monaco-editor/

JM-FRANCE, 2022-12-29:
Hi

Yes same behavior. I suppose this is the editor at the back of wokwi?

urish, 2022-12-29:
Indeed! I was hoping they might have fixed it in one of their recent releases, but unfortunately, it seems this is not the case. There is an [open feature request](https://wokwi.com/features#feature-1028025160) for improved mobile code editor, but unfortunately, it got almost no votes.

Meanwhile, it might be possible to use a different app for editing, and then copy the result to Wokwi. Not very convenient, but better than nothing I guess. I know some Android users have been doing this - see #208 for relevant discussion.

JM-FRANCE, 2022-12-29:
> Meanwhile, it might be possible to use a different app for editing, and then copy the result to Wokwi

yes that's what I do, and that's why I need an easy way to select all text from the template and delete it. At the moment on my iPhone or iPad I can only position the cursor at he end and hit backspace. 

As a short time help may be you could have a fully empty template project? it's not difficult to have a keyboard shortcut on iOS to enter
```
{
  "parts": [ { "id": "uno", "type": "wokwi-arduino-uno" } ],
  "connections": []
}
```
in the diagram

JM-FRANCE, 2022-12-29:
PS/ have you looked at https://codemirror.net?

urish, 2022-12-29:
Yes, using CodeMirror is what the [open feature request](https://wokwi.com/features#feature-1028025160) is talking about. Unfortunately, that's not a trivial change, so we'll have to patiently wait until it gets to the top.

JM-FRANCE, 2022-12-29:
OK got it. Thanks for a great product.

closed 2022-12-29 by JM-FRANCE