wokwi/wokwi-features #493: Neopixel Ring does not work using Pico Pi and Micropython

martinkoehler, 2022-12-28:
**Describe the bug**
Using the NeoPixel Ring in a simulation with a Pico Pi and Micropython does not work. The leds stay white.

**To Reproduce**
See https://wokwi.com/projects/352320936190737409

**Expected behavior**
Should behave like https://wokwi.com/arduino/projects/305569065545499202
Same python code, but other board used

**Environment (please complete the following information):**
 - Browser chrome
 - Version 108.0.5359.124

urish, 2022-12-28:
Thanks for reporting!

Upon initial investigation, connecting the [logic analyzer](https://docs.wokwi.com/guides/logic-analyzer) to the pin, it seems like the code is sending a valid WS2812 signal, even though it's all black pixels:

[image]

martinkoehler, 2022-12-28:
Note: If you change "wokwi-pi-pico" to "wokwi-esp32-devkit-v1" in diagram.json, and connect DIN to D15, it works as expected.

martinkoehler, 2022-12-28:
i also tried to use the neopixel library from micropython (https://raw.githubusercontent.com/micropython/micropython-lib/master/micropython/drivers/led/neopixel/neopixel.py) directly (as additional file, but named as `np.py` to avoid name collisions and then `from np import NeoPixel`). Unfortunately with the same result.

martinkoehler, 2022-12-29:
Indeed the Pin in the pico pi case seems to emit black pixels.
I just created two even simpler version, where a single NeoPixel is attached to `PIN(15)` and which clearly shows the problem:
The python code is:
```
from machine import Pin
from neopixel import NeoPixel
np=NeoPixel(Pin(15),1)
np.fill((0xfa,0xfa,0xfa))
np.write()
```

https://wokwi.com/projects/352400356608819201 for the ESP32 Version and a single NeoPixel
This works and the NeoPixel is white after running the Simulation
Same code and layout for the Pico Pi with Micropython
https://wokwi.com/projects/352400550196923393
Here the NeoPixel remains black.

In the first case the logic analyzer shows
[image]
(correct values (0xfa,0xfa,0xfa), which gives a white pixel)
while in the Pico Pi case one sees
[image]
(wrong values (0,0,0), which gives a black pixel)

Hope this helps for the debugging?!
Thanks for you efforts!

urish, 2023-03-22:
Moving the discussion to #551

closed 2023-03-22 by urish