wokwi/wokwi-features #495: Cannot create static array larger than 64512 bytes in custom chip code

sbdswr, 2023-01-02:
When I try to use large static arrays in custom chip code (I wanted to include a bitmap to render on the framebuffer, which was a bit over a megabyte character string), simulation dies quietly and browser console logs error:

```
2949-deb986376d7fc3ad.js:1 Uncaught (in promise) LinkError: WebAssembly.instantiate(): memory import 0 is smaller than initial 3, got 2
    at 2949-deb986376d7fc3ad.js:1:197612
    at main-7699d667fb59676b.js:1:11690
    at Object.next (main-7699d667fb59676b.js:1:11795)
    at r (_app-c5cdfa886ab9ad34.js:74:417715)
    at s (_app-c5cdfa886ab9ad34.js:74:417918)
```

The example can be found in https://wokwi.com/projects/352752191712013313. The code for the chip there is just:

```c
#include "wokwi-api.h"

char data[64513];

void chip_init() {
  char *ptr = data;
}
```

Coincidentally, removing the include lets larger arrays to be used. It'd be great to be able to squeeze a bitmap or a few there.

**Environment**
 - OS: Linux
 - Browser: Vivaldi (Chromium based)
 - Version: 5.6.2867.50

urish, 2023-01-02:
Thanks for reporting, should be fixed now!

Note: if you are testing with the same project, please modify the chip's source code (e.g. add some comment), to make sure it recompiles.

closed 2023-01-02 by urish

sbdswr, 2023-01-02:
It works. Thanks!