wokwi/wokwi-features #499: RP2040: systick apparently reporting us, not system ticks

c1570, 2023-01-04:
**Describe the bug**
On RP2040 (SDK), `systick_hw->cvr` seems to be tied to `time_us_32()` or similar.
At least at `systick_hw->csr = 0x5`, it should report system ticks/clock cycles.

While it's expected that the simulation doesn't really achieve cycle accuracy, it'd be great to have some way of finding out a ballpark estimate (and µS accuracy doesn't always cut it).

**To Reproduce**
https://wokwi.com/projects/352948898182603777

**Expected behavior**
`systick_hw->cvr` should count down system/clock cycles.

**Environment (please complete the following information):**
Chromium 108.0.5359.124, current Wokwi.com

urish, 2023-01-05:
Thanks for reporting, the issue reproduces here.

Simulator output:

```
          old-new=124  (systicks, should be >15000)
            t1-t0=124  (time_us, this should be <200)
(old-new)/(t1-t0)=1.0
```

Pi Pico output:

```
          old-new=23629  (systicks, should be >15000)
            t1-t0=194  (time_us, this should be <200)
(old-new)/(t1-t0)=121.8
```

c1570, 2023-01-29:
Changing `rp2040.clock.micros` to `rp2040.core.cycles` in https://github.com/wokwi/rp2040js/blob/61e1d2cc9953eb049f7c45f8f0d3acc39c4c55e1/src/peripherals/ppb.ts#L124 fixes the issue.

urish, 2023-01-30:
Thanks! How does it work with different values of `CSR`?

c1570, 2023-01-30:
I think CSR is broken/unimplemented? E.g., this.systickCountFlag is always false.
The change to core.cycles doesn't fix that. But at least CVR can be used to count cycles then. :) Fixing/implementing CSR is another issue/PR probably.

urish, 2023-02-05:
> I think CSR is broken/unimplemented? E.g., this.systickCountFlag is always false.

Right, it's never set to true. Seems like we never implemented this.

closed 2023-03-14 by urish