wokwi/wokwi-features #501: make LEDs more like LEDs (pinMode)

alto777, 2023-01-05:
I got burned (well, almost and not literally) by the fact that an INPUT pin can perfectly drive an LED.

So the sim worked, but it would not have IRL.

I get away without using a series current limiting resistor in the sim, so I know those are super LEDs and I know to put them in anyway if I am going to share...

but having a default pin fully illuminate an LED means Wow! LEDs have come a long way, bright on ~0.1 mA.

I don't know how they are implemented, so I hope there's an easy way to fix this.

a7

urish, 2023-01-08:
Do you mean that `INPUT_PULLUP` can drive an LED?

alto777, 2023-01-08:
Yes, it appears to do. 

It just means needing to be careful in carrying code out of the simulator - I have been impressed by the fidelity of the wokwi, and this is a case where a real error in the code (failrue to pinMode OUTPUT) wouldn't be visible in simulation.

So on my check list, a list which would ideally not need any items, is to confirm and ensure that pinModes are correct.

I did enjoy recently using the prodigious drive capability of the sim OUTUPT pins when I started playing with virtual steppers... but here too is a similar case where a noob might think s/he'd get away with wiring a stepper directly to the Arduijo board.

Neither large deals, call them observations. ;-)

a7

urish, 2023-01-09:
Thanks for pointing this out. Checked on a physical Arduino board - the pull ups can indeed drive an LED, but the LED is very dim - you can only see the light if you are looking directly into the die, or if you are in a dark room.

I can look into implementing this into the simulator, lighting up the LED very slightly when it's driven through a pull-up. One issue, though, is that the sim doesn't really know anything about resistor values - so if you are an external 10Ω pull-up, the LED will still be very dim, as if you drove it by the internal MCU pull-up. But I guess most people use the internal pull-ups (or downs) anyway, so the low-resistance external pulls are most of an edge case.

alto777, 2023-01-10:
I wonder out loud in print here if it wouldn't be OK to punt on presenting a very realistic dim LED. Just leave it off.

I haven't expected much of the analog portions of the simulator, never tried, for example, wiring an explicit voltage divider (not a slide fader or pot, that is). And there are no transistors or capacitors. I do see wonderful new to me things like the gates and the custom parts and so on. I may ever need to use them…

Thank you 10^6 for the whole thing. I will never tire of promoting use of the wokwi, and it has def been a time waster um saver I mean. ;-)

And no lie when I tell ppl who do find fault to bring things to your attention that it will be considered. Mostly the fault is not in the sim, however.

a7

urish, 2023-01-10:
> I wonder out loud in print here if it wouldn't be OK to punt on presenting a very realistic dim LED. Just leave it off.

Good point.

The only challenge now - internally, there's no difference between a pull-up resistor, and a current limiting resistor that goes between the LED and GND or VCC. It makes this change pretty complicated, and may have to wait until we have #203.  

> Thank you 10^6 for the whole thing. I will never tire of promoting use of the wokwi, and it has def been a time waster um saver I mean. ;-)

Thaks!