wokwi/wokwi-features #505: Add a keyboard shortcut for the "View Compiled Assembly Listing"

makkiato83, 2023-01-08:
At the moment there is no shortcut for this very nice feature. 

Thanks

urish, 2023-01-08:
Can you proposed a key shortcut? Please check if doesn't already exist / conflict with the browser defined keys (Ctrl+T for instance)

makkiato83, 2023-01-08:
I think ⌘§ could be an option.

Granted it is not very 'mnemonic'. On the other hand it is very fast to type and it is currently unused ( I think ).

Regarding the 'mnemonic', § is like the "S" in aSm... :-P

urish, 2023-01-08:
Hmm, I'm not sure which key is §? It probably only appears on some keyboard layouts?

makkiato83, 2023-01-08:
Uhm, sorry I should have checked. 
In my keyboard (Macbook air M2, international keyboard) it is the button on the left of "1", just below "ESC".

urish, 2023-01-08:
Oh, well, on my keyboard that is where the ` (backtick) and ~ button is.

We don't have an § key here. I guess we could go for Cmd(or Ctrl) + backtick then?

makkiato83, 2023-01-08:
Sure! That would work as well!
Thanks

urish, 2023-01-08:
Done!

closed 2023-01-08 by urish