wokwi/wokwi-features #509: Add support for SH1106 OLED chipset

freemansoft, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
Add support for SH1106 chipset for OLED displays. wokwi supports the SSD1306 OLED displays.  There are a bunch of similar OLED displays that use the SH1106 including one sold by Microcenter.  

**Describe the solution you'd like**
Add support for the SH1106 so that **MicroPython** programs that use the SSH1106 SPI / I2C displays can blah blah.

Example python code and python SH1106 driver can be seen here https://joe.blog.freemansoft.com/2023/01/inland-iic-spi-13-128x64-oled-v20.html

urish, 2023-01-29:
Thanks! Opened this request for [voting](https://wokwi.com/features?sort=newest#feature-1561379878).

urish, 2023-11-13:
We now have an experimental implementation of the SH1107: https://wokwi.com/projects/376496004605220865