wokwi/wokwi-features #510: Feature request: Upload binary file to flash instead of source code to be compiled

Semnodime, 2023-02-03:
It would be nice to have the option to compile the project locally, thus having to upload the compilation result only, to then be run in the emulated environment.

urish, 2023-02-03:
Hello, this is already supported for [Wokwi for Visual Code](https://docs.wokwi.com/vscode/getting-started). Can you give it a go and see if that works for your use case?

Semnodime, 2023-02-07:
Works, using `PlatformIO` to compile and the UI to trigger running the `wokwi` simulation.

urish, 2023-02-08:
Thanks for the feedback! Does this setup work for your use case?

Semnodime, 2023-02-08:
Everything works for me, as I just found the <kbd>F1</kbd> menu on woki.com, so I do not even need to utilize a mitm proxy to intercept and spoof traffic to the simulator.
I think the wokwi.com UI still deserves some UX improvement, but as for functionality that is already implemented (although somewhat hidden) it is all fine!

As a sidenote: Thanks for documenting the dev effort behind wokwi, the simulator, etc on youtube, From yesterday till today, I munched plenty of your videos in 5x speed and it helped me out a lot.