wokwi/wokwi-features #511: Custom analog sensor, possibly based on existing part

alextrical, 2023-02-05:
Could anyone point me in the right direction of how I should go about creating a new analogue sensor to use, ideall I'm looking to create a Thermistor that has a 100K resistance and can be set to 300C
I believe the default sensor could be a good choice if the limits could be configured, or a custom alternative can be created.

I can see the possibility to create custom chips, and have successfully used 2 in a simulation

Many thanks

urish, 2023-02-05:
Hello! 

The [Chips API](https://docs.wokwi.com/chips-api/getting-started) is the way to go. You can define the temperature input of the thermistor as a control in the chip's json file, and then change the output voltage based on the value of the control.

alextrical, 2023-02-05:
Hi, Thank you for getting back, I can confirm that I have had success  with a custom chip with Analogue output. Its looking like a great system for defining extra parts.

Keep up the great work ^_^

closed 2023-02-05 by alextrical

urish, 2023-02-05:
Great! Would love to see what you have come up with