wokwi/wokwi-features #512: No millis() function in custom part definition

alextrical, 2023-02-05:
**Describe the bug**
There looks to be no way to track the passage of time in a custom part.
When trying to call a time procedure such as 'millis();' in a part definition, the build fails with the following error
`error: call to undeclared function 'millis'; ISO C99 and later do not support implicit function declarations [-Werror,-Wimplicit-function-declaration]`

**To Reproduce**
Try to call 'millis();' or 'micros();' in a *.chip.c custom part definition 
[Simulation with issue](https://wokwi.com/projects/355817820975930369)

**Expected behavior**
I would expect 'millis();' to return the current length of time that the program has been running, and allow calculations to function that are based on passage of time

**Environment (please complete the following information):**
 - OS: Ubuntu 22.10
 - Browser: Firefox 108.0.1
 - Version: Latest web version of Wokwi

**Additional context**

urish, 2023-02-05:
There's a [get_sim_nanos](https://docs.wokwi.com/chips-api/time#uint64_t-get_sim_nanos) function that returns the current simulation time in nanoseconds. Have you looked into the documentation but missed this function?

alextrical, 2023-02-05:
That looks ideal, I definitely checked that page, but it looks like I scrolled past it. I searched all examples and docs for the keyword 'millis'

Thank you so much, I will give it a try now

urish, 2023-02-05:
Great, I've added some extra text to make the function more visible, so next time someone searches for "millis" they would find it.

alextrical, 2023-02-05:
Absolutely perfect, it is working nicely in the interface. The note you have added is nice and clear to help future users who where in my situation.

Have a great day

closed 2023-02-05 by alextrical