wokwi/wokwi-features #514: Document and support serial device

jflamy, 2023-02-06:
The `serial` device is currently unsupported and undocumented.

Opening the serial device opens a webserial connection in Chrome (or Edge).
This allows a program running com0com version 2.2 on Windows, or socat on a Mac to connect to the device.

One use for this is Firmata, which is a protocol used by several libraries.  The Firmata firmare is loaded once on the board.  A driver program is written in JavaScript, Java, Python whatever language, and talks Firmata to the Arduino.

It would be very nice if it could be supported.

urish, 2023-02-11:
Thanks! Opened this [for voting](https://wokwi.com/features?sort=newest#feature-1572861650)

urish, 2023-02-12:
This one got popular really fast - 210 votes in a day!

drf5n, 2023-04-01:
Is it the one in

 https://wokwi.com/projects/306534287811805760 

with this json:

```
{
  "version": 1,
  "author": "Uri Shaked",
  "editor": "wokwi",
  "parts": [
    {
      "type": "wokwi-arduino-uno",
      "id": "uno",
      "top": 0,
      "left": 0,
      "attrs": {}
    },
    {
      "id": "serial",
      "type": "wokwi-serial-port",
      "attrs": { "baud": "115200" }
    }
  ],
  "connections": [
    ["$serialMonitor:RX", "serial:TX", ""],
    ["$serialMonitor:TX", "serial:RX", ""]
  ],
  "serialMonitor": {
    "display": "always"
  }
}
```

urish, 2023-04-02:
Yes

rvxfahim, 2023-05-07:
I am trying to use this project, I can receive from my computer to the serial monitor (mcu) but sending Serial.println commands doesn't come to my serial monitor in the computer. Using com0com for redirecting com8 <>com7
simulator connects to com8
pc connects to com7

p.s com0com works fine with a offline simulator like Proteus.

dqj999, 2023-08-05:
> I am trying to use this project, I can receive from my computer to the serial monitor (mcu) but sending Serial.println commands doesn't come to my serial monitor in the computer. Using com0com for redirecting com8 <>com7 simulator connects to com8 pc connects to com7

Ditto here - how can you output from the simulation to com0com?

rvxfahim, 2023-08-09:
> Ditto here - how can you output from the simulation to com0com?

com0com installs 2 virtual serial ports in windows, in this case it is com 8 and com 7.
the wokwi simulator, when it wants to connect to a serial port in windows will ask chrome and chrome pops up a dialog box with available serial ports in my computer including the virtual serial ports installed by com0com. Hence, the idea is to connect wokwi with com 8 using chrome. Then you can connect to virtual com 7 using hterm/realterm and receive what wokwi is sending through com 8, you can also send anything through com 7 using hterm/realterm and wokwi should receive it from com 8.

egikaza, 2023-09-20:
Hi all,
The same with VCP ports COMx<=>COMy when connected to TERMITE serial.read and print - not working. Is it possible to read/write to/from serial port data in simulator from/to VCP port PC without SerialMonitor?
Many thanks for help and answer?

PtrTella, 2023-10-07:
Hi, is it possible to upload directly an hex file (like compiled prooject on ArduinoIDE or better PlatformIO), using serial and com0com like in offline simulator? i would be amazing

drf5n, 2023-10-07:
> Hi, is it possible to upload directly an hex file (like compiled prooject on ArduinoIDE or better PlatformIO), using serial and com0com like in offline simulator? i would be amazing

This thread is more about connecting code running in the simulator to serial in the outside world.  But within [the online simulator](https://wokwi.com/projects/new/arduino-uno) there's a "F1 / Upload Firmware and Start Simulation.

egikaza, 2023-10-09:
Hi,
I implemented Homeduino protocol on simulator https://wokwi.com/projects/376047882939385857  it works from SerialMonitor at Simulator from web page. I need to have serial connection to simulator from my pc via VCP port using any terminal application (without internal serialmonitor). Is it possible, who knows?

markaren, 2024-02-14:
> Ditto here - how can you output from the simulation to com0com?

Has anyone been able to get reading serial output from wokwi to work? 
I am able to let Wokwi recieive input, but not the other way around. I'm using com0com,

gsingh93, 2024-04-01:
Has anyone gotten this working with virtual serial ports with `socat` or any other tool on macOS or Linux? I can create the ports just fine (`socat -d -d pty,raw,echo=0 pty,raw,echo=0`), but they don't show up when Chrome asks for a serial port.

AbhishekGY, 2025-01-04:
> Has anyone gotten this working with virtual serial ports with `socat` or any other tool on macOS or Linux? I can create the ports just fine (`socat -d -d pty,raw,echo=0 pty,raw,echo=0`), but they don't show up when Chrome asks for a serial port.

I dont think the browser can see virtual ports since they work at the operating system level. You might have to write a program that bridges Wokwi's websocket and your virtual port