wokwi/wokwi-features #515: IDE / VS Code Extension / Publish as vsix or on open-vsx.org

Semnodime, 2023-02-07:
As mentioned by @OliverKeeve in this [readme](https://github.com/OliverKeefe/vscode-extensions-in-vscodium#readme), the `codium` fork (vs code without telemetry) uses a different extensions store, namely [open-vsx.org](https://open-vsx.org/).

Currently, wokwi is not available there and therefore won't be listed within the available extensions within `codium`.
The github search does not list a respective download link anywhere, so I kindly want to promote providing one.

Related to #414.

urish, 2023-02-08:
Thanks! So your use case is using the extension with `codium`, right?

Semnodime, 2023-02-08:
> Thanks! So your use case is using the extension with `codium`, right?

Correct.

Semnodime, 2023-02-08:
I circumvented the (primarily legal) trouble with switching `codium` to the microsoft vscode extension store by using vscode to install the extention, then simply moving the extention data (`.vsix`, basically a zip) into the local dir `codium` uses on my system.

Semnodime, 2023-02-08:
Stumbled upon https://pre-release.private.wokwi.com/wokwi-vscode-latest.vsix, is this still maintained?

urish, 2023-02-08:
nope, it was maintained during the private alpha period. That version is now a month+ old and does not receive updates.

urish, 2023-04-02:
I looked into this today - looks like Open-vsx.org may be shutting down in a few weeks: https://blogs.eclipse.org/post/john-kellerman/help-us-sustain-open-vsxorg

dzid26, 2023-08-17:
They haven't https://www.globenewswire.com/news-release/2023/06/27/2695137/0/en/The-Open-VSX-Registry-a-Vendor-Neutral-Community-Supported-Alternative-to-the-Visual-Studio-Marketplace-Gets-its-Own-Working-Group-at-the-Eclipse-Foundation.html

urish, 2023-08-18:
Good news - then, I'll look into this again in a couple of weeks

urish, 2023-09-05:
https://open-vsx.org/extension/Wokwi/wokwi-vscode

urish, 2023-09-05:
Would love to see some users 5-starring the project :-)

https://open-vsx.org/extension/Wokwi/wokwi-vscode

closed 2023-09-05 by urish