wokwi/wokwi-features #517: Make new projects unlisted by default

sbdswr, 2023-02-11:
As a privacy enhancement I'd like to see projects unlisted by default.

An option to make them public (i.e. not unlisted) could be possibly added to the new prompt for naming project?

urish, 2023-02-11:
Thanks for the request! 

What privacy concerns do you have with the projects being public by default?

sbdswr, 2023-02-12:
I think I've seen a description somewhere, that a public project can end up potentially on a list of featured projects. It seems that "unlisting" project still makes it available by sharing a link, and this is what I sometimes need, but at the same time I do not feel comfortable that anything I'm playing with in the emulator may someday become discoverable on Wokwi in any way, just because I saved it. Even when I decide to create a user name to be seen via https://wokwi.com/makers. I would much more like privacy by default, and opt-in for any kind of discoverability per project.

urish, 2023-02-14:
We're looking at adding an option to make projects private while saving them. We'll keep the default Public, since we want to encourage people to share their work. 

If we made it possible for only for [club members](https://wokwi.com/club) to change the default or set the projects as unlisted while saving, would that work for you?

AriellaE, 2023-02-22:
Hello @sbdswr, 
Now when saving a project you can select the privacy as you wish (for Club members)

closed 2023-02-22 by AriellaE