wokwi/wokwi-features #519: ESP32C3 emulation has a problem

Qinka, 2023-02-12:
There is a problem with the RMT emulation of the ESP32C3. When controlling WS2812 via RMT, the output signal is incomplete.

I ran the same code on esp32 and esp32c3, but got different output from neopixel ring.
[rmt-ws2812-esp32c3](https://wokwi.com/projects/356358919407847425)  and [rmt-ws2812-esp32](https://wokwi.com/projects/356453006845074433) 

I found a code from somewhere, also failed on esp32c3. [esp32-ino-ws2812](https://wokwi.com/projects/356455795151371265) and [esp32-c3-ino-ws2812](https://wokwi.com/projects/356455330387861505).

(I will test on real hardware, after it's available)

urish, 2023-02-12:
Thanks for reporting! Issue reproduced

urish, 2023-02-12:
Fixed!

closed 2023-02-12 by urish