wokwi/wokwi-features #521: RTC DS1307 error

FernandoGarcia, 2023-02-15:
Hi!

There's some bug in RTC DS1307. Setting the date to  31/5/2023 the serial monitor is showing 3/5/2023.

I have tested the same code in real hardware and it doesn't happens.

[image]

[image]

The link for simulation: https://wokwi.com/projects/356669043513165825

Best regards.

urish, 2023-02-15:
This issue happens when the current month setting in the DS1307 has less days than the requested day in the new date. For instance, if the current month is February, setting the date to anything beyond 28 may cause this issue.

urish, 2023-02-15:
Fix is up! Thanks again for the super detailed report

closed 2023-02-15 by urish

FernandoGarcia, 2023-02-15:
It's fine now, thanks!