wokwi/wokwi-features #522: RTClib error on simulate

mineirinRV, 2023-02-15:
When trying to help an OP with a problem in code using RTC, (RTClib.h),
I used Wokwi to simulate the code.

To test how it would behave when the date changes, I defined
the day as the last of the month:
I used "rtc.adjust(DateTime(2023, 5, 31, 12, 59, 40)); "
But the day appeared in the serial as the 3rd. "3/5/2023"
I tried:
"rtc.adjust(DateTime(2023, 5, 30, 12, 59, 40)); "
The day appeared in the serial as day 2. "2/5/2023"

"rtc.adjust(DateTime(2023, 5, 29, 12, 59, 40)); "
The day appeared in the serial as day 1. "1/5/2023"

But with "rtc.adjust(DateTime(2023, 5, 28, 12, 59, 40)); "
The day appeared on the serial as the 28th. "28/5/2023"

And this occurs with any month I select.

Am I doing something wrong, are there problems with my PC or is there a "bug" on the Wokwi website?

Thank you all.
Simulation link:

[wokwi.com 1](https://wokwi.com/projects/356669043513165825)

[Error_RTC - Wokwi Arduino and ESP32 Simulator 1](https://wokwi.com/projects/356669043513165825)
Run IoT and embedded projects in your browser: ESP32, Arduino, Pi Pico, and more. No installation required!

To make it easier, I simplified the code in the simulation.

#include "RTClib.h" 
RTC_DS1307 rtc;
//-----------------------------------------------------------------------------
void setup () {
  Serial.begin(115200);
  if (! rtc.begin()) {
    Serial.println("DS1307 not  found");
    while (1);
  }
  rtc.adjust(DateTime(2023, 5, 31, 12, 59, 40)); //(year), (month), (day), (hour), (minute), (second)
  delay(100);
}
//-----------------------------------------------------------------------
void loop () {
  DateTime now = rtc.now(); 
  Serial.print(now.day()); 
  Serial.print('/');
  Serial.print(now.month(), DEC); 
  Serial.print('/'); 
  Serial.println(now.year()); 
  delay(2000);
}

urish, 2023-02-15:
Thanks for reporting - it might be a duplicate of #521, which we just uploaded a fix for a minute ago. 

Can you please try again and let me know if the problem is still there?

mineirinRV, 2023-02-15:
Ok, 
I tested it now and it worked correctly.

Thanks

closed 2023-02-15 by urish