wokwi/wokwi-features #523: Specific partitions.csv in vscode with wokwi.toml

Qinka, 2023-02-15:
How can I specific partitions.csv in vscode with wokwi.toml?

I got an error in VSCode, which is the same as when I used wokwi-server without the parameter `-p partitions.csv`.

urish, 2023-02-15:
Partitions.csv is not supported in VS Code. Instead, you should directly use the partition-table.bin file generated by whatever build tool you use.

Wokwi for VS Code doesn't support (yet) multiple .bin files for ESP32, so there are basically two options:
1. Merge all binary files (bootloader, partition table, etc) into a single one using [esptool merge_bin](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/esptool/basic-commands.html#merge-binaries-for-flashing-merge-bin)
2. Generate a .uf2 file that includes everything (bootloader, partition table, app). If you are using esp-idf, `idf.py uf2` will generate such file for you. You can then point the `firmware` configuration in `wokwi.toml` to that UF2 file.

Qinka, 2024-08-03:
I think adding something `partition_tabel="path/to/partition.bin" in `wokwi.toml` is a good idea.

urish, 2024-12-01:
If you are using idf.py to build your project, you can point wokwi.toml at build/flasher_args.json to automatically load the partition table, as explained here: https://docs.wokwi.com/vscode/project-config#esp-idf-support